teleshoes/tpacpi-bat #52: Force Charge

wasutton3, 2015-03-29:
Is there a way to force the system to charge? As in, the last discharge wasn't enough to drop below the threshold, but the next time I'll need it to be on battery, it won't have enough charge. Any solution? Similar to the charge-inhibit, but the exact opposite, a "force-charge" option (to the respective threshold at least).

teleshoes, 2015-03-30:
just set the threshold and it will start charging, no?

wasutton3, 2015-03-30:
It would, but then you have to fiddle with the threshold and make it change
back. maybe a toggle to set the start charge threshold to 0%? then an
interlock with the charge inhibit so it doesn't lose its mind.

teleshoes, 2015-03-30:
im not following you. this is what you want to do, right?
```
old=`tpacpi-bat -g ST 1`
tpacpi-bat -s ST 1 0
bat=0
while [ $bat -lt 100 ]; do
  now=`cat /sys/class/power_supply/BAT0/energy_now`
  full=`cat /sys/class/power_supply/BAT0/energy_full`
  bat=$((now * 100 / full));
  sleep 3600
done
tpacpi-bat -s ST 1 $old
```

to be clear, there is nothing in the thinkpad acpi interface for doing this directly. that being said, the api almost certainly exposes enough functionality for doing whatever it is that you want with a simple script

linrunner, 2015-03-30:
The *only* way is to achieve this is to temporarily set the lower threshold above the charge level.

You may use [TLP](http://linrunner.de/en/tlp/docs/tlp-linux-advanced-power-management.html) to manage your thresholds (via tpacpi-bat of course). It has the command 

    tlp chargeonce

exactly for this use case.

closed 2015-03-30 by teleshoes