teleshoes/tpacpi-bat #53: Possible to get this to work with IdeaPad Edge 15?

lynema, 2015-04-03:
The Windows install has a tool to set the charge values.  I'm trying to find the ACPI values to set to get this working in Linux.

/sys/class/power_supply/BAT0/device/path = \_SB_.PCI0.LPCB.EC0_.BAT0
Trying to "echo \_SB_.PCI0.LPCB.EC0_.BAT._BIF   > /proc/acpi/call" always gives a AE_BAD_PARAMETER error.  Not sure where I'm wrong there.
I've dumped the ACPI to dsdt.dsl.  This is the only place HKEY shows up
    Scope (\)
    {
        Mutex (MSMI, 0x00)
        Method (WSMI, 1, NotSerialized)
        {
            Store (Arg0, SFUN) /* \SFUN */
            Store (TSMI, SSMP) /* \SSMP */
        }

        Method (HKEY, 1, NotSerialized)
        {
            TRPS (0x14, Arg0)
        }

Here is the Battery
        Name (B0ST, Zero)
        Device (BAT0)
        {
            Name (_HID, EisaId ("PNP0C0A") /* Control Method Battery */)  // _HID: Hardware ID
            Name (_UID, One)  // _UID: Unique ID
            Method (_PCL, 0, NotSerialized)  // _PCL: Power Consumer List
            {
                Return (_SB) /* \_SB_ */
            }

Of course, there is more to it then that.  Please let me know if it is possible to support this machine and what I can do to help.

teleshoes, 2015-04-03:
pretty sure this ACPI interface was used only for certain thinkpads. it is unlikely that the same interface is exposed on non-thinkpads.

its possible that there is a similar (completely undocumented) ACPI interface for ideapads lurking somewhere that you can reverse engineer. good luck!

closed 2015-04-03 by teleshoes

linrunner, 2015-04-03:
@teleshoes: the *really* interesting part – for ThinkPads – is: one may directly retrieve the aslBase from /sys/class/power_supply/BAT0/device/path instead of iterating (brute force) through all possible alsBases ... nice approach.

teleshoes, 2015-04-03:
oh shit! i was just skimming and didnt notice.
awesome, thanks @lynema! {and @linrunner for actually reading}

ill post a release in a few minutes.

teleshoes, 2015-04-03:
see #54 for asl-base-auto-detect

back on the actual topic of this issue, on thinkpads you need to remove the trailing underscores from each segment, so maybe try ```\_SB_.PCI0.LPCB.EC0.HKEY``` ?

linrunner, 2015-04-04:
@lynema: could you try and show

    tpacpi-bat -v -g  ST 1

with the new version?

lynema, 2015-04-05:
[root@edge15-N lyz]# ./tpacpi-bat -v -g  ST 1
no AC device path to read ASL base from: /sys/class/power_supply/AC/device/path

Actual device is here:
[root@edge15-N lyz]# cat /sys/class/power_supply/ADP1/device/path
\_SB_.PCI0.LPCB.EC0_.ADP1

After edit to set the location correctly
[root@edge15-N lyz]# ./tpacpi-bat -v -g  ST 1
Error: AE_NOT_FOUND for ASL base: \_SB.PCI0.LPCB.EC0.ADP1
Error: AE_NOT_.FOUND at ./tpacpi-bat line 402.
Makes sence as it is missing an underscore (_SB_)

Side note: I did figure out the way to make a working call to the battery:
echo "\_SB_.PCI0.LPCB.EC0_.BAT0._BIF" > /proc/acpi/call

I examined some of the output from the Thinkpad ACPI dumps.  The IdeaPad is _way_ different.  The methods under BAT0 all all status calls _STA, _BIF, _BST.  Not sure if having the right ASL base is going to help as it seems like there is something else going on.  I have no idea how the power manager for windows is able to set the threshold.  The threshold persists across a reboot to my Linux partition.

teleshoes, 2015-04-05:
```\_SB.PCI0.LPCB.EC0.BAT``` would be correct, if the interface was there, and not ```\_SB_.PCI0.LPCB.EC0.BAT```

ill fix it to use BAT0 instead of AC tho

linrunner, 2015-04-05:
> ill fix it to use BAT0 instead of AC tho
Hardcoding BAT0 will break compatibility with some ThinkPads having BAT1 instead of BAT0 as first battery. I suggest you iterate over BAT0, BAT1 and take the first occurence.

teleshoes, 2015-04-05:
sigh, forgot. ill check [AC, BAT0, BAT1, ADP0, ADP1]. at least all im doing is a simple stat on a handful of files instead of making up to 9 ACPI calls; still a large improvement.

teleshoes, 2015-04-05:
also, we should really issue #54 for this, fairly OT

willsALMANJ, 2015-04-05:
I was just trying to figure this out for my new computer as well. It's a Lenovo Z50-70. It's not branded as ThinkPad or IdeaPad, so I haven't been sure where to look for Linux support information. I followed the suggestions in this and a few other closed issues to get the DSDT.dsl file (I put it in [this gist](https://gist.github.com/willsALMANJ/c36dc6a3acd80d8e17b7) if you want to see it). There is a "Lenovo IdeaPad" string in the BAT0 device section, so perhaps it uses the same interface as the IdeaPad.

There is a utility in the Windows installation that sets the charging thresholds. I have found that Linux initially respects the thresholds but over time the computer switches back to fully charging. I haven't tested this enough to find if something specific triggers the change.

I saw the previous suggestion that if the interface differs from ThinkPad that it's not worth bothering with, but is there anything else I can try that is not too consuming or dangerous? Should the interface for controlling the charging thresholds be in the DSDT.dsl file, or could it be left out if Linux does not know how to use it? When I ran `iasl` to create the DSDT.dsl file, I got the warning "iASL Warning: There were 11 external control methods found during disassembly, but only 3 were resolved (8 unresolved)."

Could anything be learned from the Windows utility's files?

teleshoes, 2015-04-05:
yea, there isnt necessarily an acpi interface, and even if there is, it is unlikely you will be able to reverse engineer it. i personally lack the ability to do this.

as for what makes it reset, ive seen this happen at a fixed number of reboots (9 in my case). it could also be merely removing the battery, dropping beneath a critical threshold, or any number of other things.

id say your best bet is to stop caring and buy a replacement battery in a year. and maybe consider getting an open firmware device or at least use a vendor that doesnt "recommend windows 8" and hate GNU/Linux. good luck...

willsALMANJ, 2015-04-06:
Thanks for the response. I'll try not to care about this. I wouldn't mind spending time figuring it out if it were possible (even if it weren't monetarily worth the time investment) because I like learning about the inner workings of things, but it seems like the efforts would be fruitless. This was the first computer I bought with the intention of running Linux as the primary OS. I'll have to consider the open vendors more closely next time. I hadn't realized the distinction between ThinkPad and IdeaPad for Linux before, and I thought I was doing pretty well otherwise in terms of choosing a machine for which all the components had Linux drivers. The cheapest equivalent computers that I could find from more freedom-loving vendors like system76 had a 50% markup compared to the one I bought...so freedom comes at quite a steep price (and those computers might not have had the ability to control the battery charging).

teleshoes, 2015-04-06:
honestly, thinkpad is probably still among the best for linux support, but lenovo has been getting steadily worse in this regard {and, recently, worse in every other regard} since they bought thinkpad from IBM, and its depressing. there isnt really anyone else out there, with folks like zareason and system76 being the mostly-symbolic exceptions