teleshoes/tpacpi-bat #56: Problems on X121e

quivocal, 2015-04-22:
Something doesn't seem to be working right here.

    ~ $ sudo tpacpi-bat -v -g ST 1                                          
    Error: AE_NOT_FOUND for ASL base: \_SB.HKEY
    Error: AE_NOT_FOUND at /usr/bin/tpacpi-bat line 409.  

    ~ $ cat dsdt.dsl | grep \\\\_SB\.PCI.*HKEY -o | uniq
    \_SB.PCI0.LPCB.EC.HKEY

    ~ $ cat /sys/class/power_supply/BAT1/device/path                         
    \_SB_.BAT1

    [ 6471.594319] acpi_call: Cannot get handle: Error: AE_NOT_FOUND

dmesg is full of those errors. If I override $aslBase to the value extracted from dsdt.dsl above I get AE_BAD_PARAMETER error instead.

I also have a bunch of acpi related errors in dmesg;
```
[    3.655200] ACPI Warning: SystemIO range 0x0000000000000428-0x000000000000042f conflicts with OpRegion 0x0000000000000400-0x000000000000047f (\PMIO) (20141107/utaddress-258)
[    3.655215] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver
[    3.655222] ACPI Warning: SystemIO range 0x0000000000000540-0x000000000000054f conflicts with OpRegion 0x0000000000000500-0x0000000000000563 (\GPIO) (20141107/utaddress-258)
[    3.655231] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver
[    3.655235] ACPI Warning: SystemIO range 0x0000000000000530-0x000000000000053f conflicts with OpRegion 0x0000000000000500-0x0000000000000563 (\GPIO) (20141107/utaddress-258)
[    3.655243] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver
[    3.655246] ACPI Warning: SystemIO range 0x0000000000000500-0x000000000000052f conflicts with OpRegion 0x0000000000000500-0x0000000000000563 (\GPIO) (20141107/utaddress-258)
[    3.655254] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver
[    3.655257] lpc_ich: Resource conflict(s) found affecting gpio_ich
[    3.719980] ACPI Warning: SystemIO range 0x000000000000efa0-0x000000000000efbf conflicts with OpRegion 0x000000000000efa0-0x000000000000efaf (\_SB_.PCI0.SBUS.SMBI) (20141107/utaddress-258)
[    3.719995] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver
```
```
 ~ $ sudo dmidecode -s bios-version && sudo dmidecode -t11                
[sudo] password for quiv: 
8QET58WW (1.19 )
# dmidecode 2.12
# SMBIOS entry point at 0xbae65000
SMBIOS 2.6 present.

Handle 0x0034, DMI type 11, 5 bytes
OEM Strings
	String 1: This is the Intel HuronRiver CRB Platform
```

It also seems that my battery is now holding a charge at a constant level while plugged into the ac but will not charge above that level. I guess I must have some how accidently set the ST/SP parameters. Maybe I will need to take the batery out to reset it to the 96/100 default.

Not sure what else to do, something seems to be going wrong somewhere. I find it particularly disconcerting that it seems to pick up an empty path in /sys/power.

teleshoes, 2015-04-22:
yep that is weird. if setting the ASL base to whats in the dsdt doesnt work, then the problem is not in tpacpi-bat.

removing the battery and replacing will probably reset it. if not, if you remove the battery, remove the AC, and hold the power button for 10s, it will remove all bios configs.

good luck

quivocal, 2015-04-22:
Removing the battery and replacing it seems to have reset it as it's once again charging. When I get some more time I will look into the issue some more, though as you say I guess it probably isn't a problem with tpacpi-bat. Feel free to close this.

Thanks for your assistance.

teleshoes, 2015-04-23:
no problem

closed 2015-04-23 by teleshoes