teleshoes/tpacpi-bat #57: Charging thresholds appear to remain, even after acpi_call and tpacpi-bat uninstall, then full distro reinstall

brendes, 2015-07-18:
**TL;DR:** Battery charge behavior appears to adhere to previously set thresholds, even after uninstall of acpi_call, tpacpi-bat, and finally complete distro re-install (including format of root partition). Running Debian Jessie.

**Description:** Shortly after installing tpacpi-bat and setting charging thresholds (40% and 90%) on Debian Testing, I started to experience strange behavior from my Thinkpad t440 while resuming from hibernation (hung indefinitely during restore phase). In order to narrow down the issue, I first uninstalled tpacpi-bat and acpi_call, then after seeing no change eventually wiped my root partition and installed Debian Stable.

The hibernation problem is gone, but strangely, running `acpi` returns "Battery 0: Unknown, 90%", exactly as it would have if I was running tpacpi-bat with my threshold at 90%. I removed the battery for a few minutes, then re-inserted it, with no change.

My knowledge of hardware is extremely limited. Does my battery somehow "remember" these thresholds? If so, how can I completely reset it without tpacpi-bat? Apologies if I am missing something obvious.

teleshoes, 2015-07-18:
yea its stored in the bios. easiest fix is to reinstall tpacpi-bat and simply unset the thresholds. {presuming that tpacpi-bat didnt cause your hibernation troubles, which it almost certainly didnt}

{some combination of removing the battery and restarting might also trigger the reset, or killing the cmos by unplugging, removing battery, and holding power button for 30s}

also, may i recommend debian sid? testing is generally unusable IMO, while the 'unstable' distro is actually much more reliable. ive been using it via siduction for some years now. aside from having to pay careful attention to the updates, ive found it to be surprisingly reliable, for a bleeding-edge rolling distro.

closed 2015-07-18 by teleshoes

brendes, 2015-07-18:
Aha, so it's in the BIOS. So I'm guessing that if it were responsible for the hibernation problem then I'd likely still be seeing it, as my battery is behaving the same.

I'm actually pretty content with Stable for the time being, but if I get antsy again I'll look at Sid instead of Testing. There was a kernel upgrade around the time I installed tpacpi-bat, so I wasn't sure what was causing my problem, and that's why I decided to just reinstall everything.

Thanks a lot.

teleshoes, 2015-07-18:
no problem