teleshoes/tpacpi-bat #62: [Question] Thinkpad T430 don't stop to charge, running Linux Mint 17.2

DarthBlack, 2015-09-23:
My battery is fully charged and the AC adapter in plugged-in now and when I run the following commands. My problem is that the laptop don't stop to charge after I have run the commands and rebooted. The computer is a `Lenovo Thinkpad T430` running `Linux Mint 17.2`, docked to a `Lenovo Thinkpad 4338` docking station. I'm only using a main battery, no secondary. I installed the package, and the dependencies, from the PPA repo.

I have run the commands
```
sudo tpacpi-bat -s ST 0 40
sudo tpacpi-bat -s SP 0 80
sudo tpacpi-bat -s IC 0 1
```
And the values are set because then I run the "get" commands they return
```
sudo tpacpi-bat -g ST 1 --> "40 (relative percent)"
sudo tpacpi-bat -g SP 1 --> "80 (relative percent)"
sudo tpacpi-bat -g IC 0 --> "yes (unspecified min)"
```

write-your-own-os, 2015-09-23:
Now then I have used the laptop without charger it's down to 78%, but then I plugin the AC adapter it wont charge.

teleshoes, 2015-09-24:
..theres a ppa for tpacpi-bat? anyway, im not clear on the problem exactly.

couple quick things, just to make sure:
1) inhibit charge doesnt force the battery to discharge. it stops charging, and continues to run off a/c power.
2) maybe you want force discharge?
3) the battery will never charge while inhibit charge is on. turn it off to see the battery charge.

closed 2015-09-26 by teleshoes