teleshoes/tpacpi-bat #63: Support for ideapad ? Lenovo z500

sooorajjj, 2015-10-09:
Hey , will tpacpi work for lenovo z500 , iam using LinuxMint-17.2 Lenovo z500 fixed Nvidia gpu issues (installed bbswitch successfully ), also fixed Screen backlit issue iam only left with this battery issue Battery indicator is not stuck it drops charge when i remove power cord but charges to only 59% and wont go any further

sooorajjj@Treako ~/Desktop/AOSP $ acpi -i
Battery 0: Unknown, 59%
Battery 0: design capacity 3092 mAh, last full capacity 2905 mAh = 93%

sooorajjj@Treako ~/Desktop/AOSP $ acpi -V
Battery 0: Unknown, 59%
Battery 0: design capacity 3092 mAh, last full capacity 2905 mAh = 93%
Adapter 0: on-line
Thermal 0: ok, 54.0 degrees C
Thermal 0: trip point 0 switches to mode critical at temperature 127.0 degrees C
Thermal 0: trip point 1 switches to mode hot at temperature 127.0 degrees C
Cooling 0: x86_pkg_temp no state information available
Cooling 1: intel_powerclamp no state information available
Cooling 2: Processor 0 of 10
Cooling 3: Processor 0 of 10
Cooling 4: Processor 0 of 10
Cooling 5: Processor 0 of 10

sooorajjj@Treako ~/Desktop/AOSP $ cat /sys/class/power_supply/BAT1/status
Unknown
sooorajjj@Treako ~/Desktop/AOSP $ cat /sys/class/power_supply/BAT1/uevent
POWER_SUPPLY_NAME=BAT1
POWER_SUPPLY_STATUS=Unknown
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_TECHNOLOGY=Li-ion
POWER_SUPPLY_CYCLE_COUNT=0
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=14400000
POWER_SUPPLY_VOLTAGE_NOW=14818000
POWER_SUPPLY_POWER_NOW=655350000
POWER_SUPPLY_ENERGY_FULL_DESIGN=45820000
POWER_SUPPLY_ENERGY_FULL=43050000
POWER_SUPPLY_ENERGY_NOW=25620000
POWER_SUPPLY_CAPACITY=59
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_MODEL_NAME=PABAS0241231
POWER_SUPPLY_MANUFACTURER=LENOVO 
POWER_SUPPLY_SERIAL_NUMBER=41167
sooorajjj@Treako ~/Desktop/AOSP $ 

sooorajjj@Treako ~/Desktop/AOSP $ upower --enumerate 
/org/freedesktop/UPower/devices/line_power_ACAD
/org/freedesktop/UPower/devices/battery_BAT1
sooorajjj@Treako ~/Desktop/AOSP $ upower -i /org/freedesktop/UPower/devices/battery_BAT1
  native-path:          BAT1
  vendor:               LENOVO
  model:                PABAS0241231
  serial:               41167
  power supply:         yes
  updated:              Sunday 04 October 2015 02:17:32 PM IST (27 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               charging
    energy:              25.62 Wh
    energy-empty:        0 Wh
    energy-full:         43.05 Wh
    energy-full-design:  45.82 Wh
    energy-rate:         0.0171658 W
    voltage:             15.009 V
    percentage:          59%
    capacity:            93.9546%
    technology:          lithium-ion

linrunner, 2015-10-09:
Hi, Ideapads are not supported.

sooorajjj, 2015-10-09:
@linrunner trying to add support

teleshoes, 2015-10-09:
what he said. sry.

closed 2015-10-09 by teleshoes