teleshoes/tpacpi-bat #68: "Dynamic" threshold are wrong / misleading

nguillaumin, 2016-05-08:
Hi,

In the systemd "dynamic" configuration:

```
# Start charging threshold, in format "lower-bound higher-bound".
STARTTHRES="0 40"

# Stop charging threshold, in format "lower-bound higher-bound".
STOPTHRES="0 80"
```

Unless I'm missing something, I think "lower-bound" / "higher-bound" is wrong, the first number is the battery number (1, 2, or 0 for either/both) and the second number is the actual threshold. I believe that's the case because they're passed down to `tpacpi-bat` in the following fashion in the unit file: `/usr/bin/tpacpi-bat -s ST $STARTTHRES`, resulting in `/usr/bin/tpacpi-bat -s ST 0 40` for instance.

So in practice there's nothing "dynamic" about these threshold, it's just using a separate config file rather than hardcoding the thresholds in the systemd unit file.

I suggest either renaming it, or perhaps just remove the hardcoded example and only keep the one with the separate config file as it's probably better anyway to not have to modify the unit file to change the thresholds. Also, perhaps separating the battery number in its own variable would make it easier to understand:

```
# Battery number (1 for main, 2 for secondary, 0 for either/both)
BATTERY=0

# Start charging threshold, in %
STARTTHRES=40

# Stop charging threshold, in %
STOPTHRES=80
```

Then in the unit file:

```
ExecStart=/usr/bin/tpacpi-bat -s ST $BATTERY $STARTTHRES
ExecStart=/usr/bin/tpacpi-bat -s SP $BATTERY $STOPTHRES
```

Cheers,

Nico

teleshoes, 2016-05-08:
1) yep, the comment is wrong, it should say in format "BAT THRESHOLD". thanks!

2) `using a separate config file rather than hardcoding the thresholds in the systemd unit file` is what dynamic means in this context; you can configure the thresholds without modifying the service file.

3) i dont maintain or use the systemd service scripts, but this should be a straightforward fix

teleshoes, 2016-05-08:
thanks! 1fdbc80

closed 2016-05-08 by teleshoes

nguillaumin, 2016-05-08:
Excellent, thanks for the quick fix! :+1: