teleshoes/tpacpi-bat #70: Apparently not functioning correctly on ThinkPad L560

gde061, 2016-10-09:
Hi - This is my first post - I am not a developer, just a linux enthusiast.  I recently upgraded from a Thinkpad E530 to L560.  

I installed tpacpi-bat and executed from the terminal with thresholds of 35 and 80 respectively for start and stop thresholds.  For this I used the file /etc/battery-charge-levels and the script: battery-set-normal-charge-levels in the tpbat-utils-acpi package. 

Checking the values with tpacpi-bat -g -ST 1 returns 35 and tpacpi-bat -g -SP 1 returns 80.    This seemed like a nice improvement from the e530, which gave "unknown" values like 178 instead of the actual set values, but it did work right. 

With the L560, however, the threshold is not functioning correctly.  When for example discharged to 95% it should not start charging.   The observed behavior I get is when plugged in at 95%, the battery charges up to 99% as displayed in the Battery Charge Monitor 1.12.1 widget.

The machine has UEFI BIOS N1HET54W 1.18.

teleshoes, 2016-10-10:
if tpacpi-bat returns the right values and doesnt work, there is nothing to be done about it on tpacpi-bat's end.
coupla suggestions:
1) try battery 2, jic
2) try inhibit-charge and force-discharge commands just to make sure that youre having some sort of effect with tpacpi-bat
3) if inhibit-charge comand works, then you can write a script that runs every 60s, checks the battery level, and inhibits charging when above 80, or turns of inhibit charge

good luck!

closed 2016-10-10 by teleshoes

gde061, 2016-10-10:
Great suggestions... I wouldn't have thought of that.  

One further thing to report just in passing - my particular machine is set up in dual boot (Win 7).  Since my understanding is that settings made in Windows with the ThinkPad battery / power manager utility  should be carried forward regardless of which OS is loaded... until overriden ... I changed the thresholds in the Windows with the Lenovo  app and rebooted into Ubuntu and range the battery-get-charge-levels script, which returned the new thresholds I had set in Windows.  However the charging policy still does not honor them in that the battery charged right up to 100% again.  

Next step trying inhibit-charge will be tomorrow.   Thanks!

linrunner, 2016-10-10:
If the thresholds written with the Lenovo app in Win7 have no effect then it's a hardware issue. I'd contact Lenovo support on that.

BIOS 1.18 is the newest.

gde061, 2016-10-10:
UPDATE:  so last night I let the battery charge up in Windows to see whether the thresholds were working there.  They are not.  I tried several versions of the Lenovo power manager application and driver to no avail.  

I went back to ubuntu and tested inhibit-charge and force-discharge.   They ARE working, although the <acbreak> flag for force-discharge seems to be working backwards: 1 is preventing it from discharge when AC is ATTACHED.  0 seems to work normally.

Like I said, I'm not a developer, but I may give a try to write that script you suggested.  It seems like I'm going to have better control over the charge cap in Linux than I do in Windows using the Lenovo provided utilities.

linrunner, 2016-10-11:
Maybe it is necessary to discharge the battery *once* below the lower threshold of 35% --> http://linrunner.de/en/tlp/docs/tlp-faq.html#startthreshnotapply

gde061, 2016-10-13:
> Maybe it is necessary to discharge the battery once below the start threshold of 35% --> http://linrunner.de/en/tlp/docs/tlp-faq.html#startthreshnotapply

Looks like that was it.  Drained the battery down to 15% and then tested with the original basic thresholds...  problem appears to be resolved!