teleshoes/tpacpi-bat #72: Tag version / Release

damadmai, 2017-01-03:
Can you please tag the current HEAD with a version label / release eg. v3.1
I would need that for making a package.

Thanks for merging https://github.com/teleshoes/acpi_call/pull/2 so promptly.
I collected the contributions from various authors here: https://github.com/damadmai/acpi_call

teleshoes, 2017-01-03:
cool, done.
would you like me to make my master of `acpi_call` even with yours, or shall i modify `tpacpi-bat`'s install script to point to yours? (i tested yours and it works fine)

teleshoes, 2017-01-03:
to be clear, i cannot maintain the DKMS stuff as i wont be using it personally, but im happy to accept patches for it, or to offload the maintenance on someone else (aka you).

i still maintain and use both `tpacpi-bat` and `acpi_call` for kernel regressions and bugfixes, but they've been more or less stable for the last few years.
truthfully, tpacpi-bat should be rewritten as a kernel module, or if someone felt really ambitious, merged into thinkpad_acpi or something and mainlined, but i wont bother as long as acpi_call continues to work.

damadmai, 2017-01-03:
Maybe just stay with `tpacpi-bat` as is and update `acpi_call`. When there will be future changes I will inform you.
Yeah, it would be cool if it were merged into `thinkpad_acpi`, I wonder why nobody has done that already. :)

linrunner, 2017-01-04:
> Yeah, it would be cool if it were merged into thinkpad_acpi

Actually there was an attempt, but it got stuck: https://github.com/julian-klode/linux/commits/linux-4.2

teleshoes, 2017-01-04:
ah, i didnt know that. oh well, would have been cool

closed 2017-01-04 by teleshoes

damadmai, 2018-10-19:
Can you please tag the current HEAD with a version label / release eg. v3.2.
Thanks!