teleshoes/tpacpi-bat #73: Hardware compatibility

gi1242, 2017-01-06:
Hi There,

Thanks for writing this! It makes my life much better.

I know (from reading the issues) that you have no interest in maintaining a hardware matrix. However the information is hard to find, and I was really slow to try this out without finding someone elses success story first...

I created a wiki page posting my succesful test result. Perhaps you would consider linking to it in the README, and encouraging users to post similar success / failures.

GI

gi1242, 2017-01-06:
PS: If it wasn't clear from my previous message, I created a wiki page ON THIS PROJECTS GITHUB.

teleshoes, 2017-01-07:
heh. i would have preferred it live somewhere that DOESNT imply that i maintain it, but i stuck disclaimer on it instead. i added my laptops and renamed it to supported-hardware.

if you or anyone else wants to comb through the closed PRs/issues and find other people who claim their thinkpad works, thatd be awesome.

there are a few different levels of compatibility, btw. (see the other open issue about some machines reading the wrong values) no idea how the wiki should reflect this, but i think im done editing it personally. ;)

closed 2017-01-07 by teleshoes

gi1242, 2017-01-07:
On Sat, Jan 07, 2017 at 12:27:32PM -0800, Elliot Wolk wrote:

> heh. i would have preferred it live somewhere that DOESNT imply that i
> maintain it, but i stuck disclaimer on it instead.

Ha! Yes indeed. But at least this should be easy enough for people to
find and contribute to.

Thanks for including it!

GI

-- 
'Hospitality' -- Making your guests feel at home, even if you wish they
were.