teleshoes/tpacpi-bat #76: Issue with Battery Charge Thresholds on Thinkpad X240

DanielWhite95, 2017-01-30:
I'm using tpacpi-bat on Fedora 25 (kernel 4.9.5), but I can't change anymore the charging thresholds for bay-battery(I've set them one time to 60%-80% and now my laptop won't charge over 79%). I've tried to unset them or to set them to different values, but it still stop on 79%. 
tpacpi-bat seems to work fine:  
# tpacpi-bat -g ST 2
0 (default)
# tpacpi-bat -g SP 2
0 (default)
# tpacpi-bat -g IC 2   
no
# tpacpi-bat -g FD 2
no

NOTE: I've also tried to full discharge both batteries, but it didn't work too.

teleshoes, 2017-01-30:
-in case you havent, try seting a much LOWER stop (30% or something) and see if it works. if it does, then its probably not SP
-try 0 (and 1 i suppose) instead of 2, and then take out the main battery and try it with 0 again
-try setting all the properties/combinations, hey who knows
-try resetting the cmos (remove all batteries and ac, press and hold power button for 30s and release, press and release power button for 1s twenty times)
-maybe something like tlp is runing tpacpi-bat behind your back, or maybe x240 has some other mechanism for setting it
-or your battery could be defective, bad cell or something. i dunno

anyway, i have no real idea why it should stop working. good luck!

DanielWhite95, 2017-02-01:
It's not SP, because for values under 80 it works fine. I will try resetting the CMOS. Is it a safe procedure or is there some warning about it?

teleshoes, 2017-02-01:
it will reset all bios settings (probably). it may or may not reset your on-board system clock. aside from that, it should cause no harm.

HOWEVER, i do not have specific knowledge for your thinkpad model's hardware, so i can make no promises about safety beyond saying that this is a procedure i do all the time on many varieties of laptops and have never had anything bad happen. id advise you to do some brief research first, and go for it.

it has a small, but non-zero chance of fixing your problem. good luck!

DanielWhite95, 2017-02-07:
Nothing... I tried to fix the issue resetting BIOS settings, but it didn't work. :(

teleshoes, 2017-02-07:
dang sorry. my last idea is try a linux bootable usb stick. maybe you have some software that is doing it behind your back.

DanielWhite95, 2017-02-11:
Do you suggest me any live distro? Because no one that I know is shipped with acpi_call module or developer tools to build that.

linrunner, 2017-02-11:
Ubuntu Live System with

    sudo apt-get install acpi-call-dkms

will do. But i don't think this will make any difference.

Did you try to remove / re-insert the battery?

Did you try

> remove all batteries and ac, **press and hold power button for 30s** and release

yet? To disable the internal battery (BAT0) enter the BIOS --> Config --> Power.

teleshoes, 2017-02-11:
@DanielWhite95 try the emergency-reset thing linrunner suggests.
so the problem must be one of these:
1) user software (e.g.: tlp) is calling tpbatstat to set the stop at 80
2) some OTHER interface aside from this acpi one exists in x230 that the bios is using that was set from windows or the bios
4) the battery has a broken cell or is misreporting voltage or some such

i was saying you can eliminate (1) as a possible cause with a linux-live-usb.
(1) is already not very likely though because a lower stop threshold is honored.

if you do a full bios reset and the problem persists, its probably not (2).

teleshoes, 2017-02-11:
@DanielWhite95 
as for helping with devel, all patches are welcome! ill merge any PRs unless i have a good, specific reason, and then you could just fork it anyway.

if you wanted to, say, rewrite this as a standalone dkms kernel module that doesnt depend on acpi-call, that would be WONDERFUL. (exposing the controls as file handles in `/sys`, or whatever)

if you want to add features like polling system settings and automatically setting things, etc, etc, i would suggest exploring tlp and possibly making the contributions there. (tpacpi-bat should just be a wrapper around the acpi interface and nothing more)

HOWEVER, all the functional stuff (like controlling threshold behaviour, etc) is very unlikely to be possible to change. tpacpi-bat can only use what lenovo gives us

DanielWhite95, 2017-02-12:
> Did you try

    remove all batteries and ac, press and hold power button for 30s and release

@linrunner thank you so much, finally I full-charge my secondary battery.

teleshoes, 2017-02-12:
yay congrats!

closed 2017-02-12 by teleshoes