teleshoes/tpacpi-bat #77: tpacpi-bat cause T470s' fun runs at maximum speed after resuming from suspend to ram

cnzhx, 2017-04-01:
As stated in the title, the CPU fan in T470s with i7-7600U will run at a high speed after resuming from suspend when tpacpi-bat is installed while the service is not enabled using systemctl. This problem disappears after removing its tpacpi config in `/etc/conf.d/tpacpi` and service file in `/usr/lib/systemd/system/tpacpi.service` (this service has not been enabled though systemctl).

P.S. Does TLP use tpaci-bat as a component?

teleshoes, 2017-04-01:
1) tlp does use tpacpi-bat, yes
2) the only thing this project addresses is the tpacpi-bat perl script, period. the services/etc were written by other folks who really wanted a place to host them. i do not use them personally, and i do not provide support for them.
3) if you find a bug in tpacpi-bat itself, please report it.

closed 2017-04-01 by teleshoes

linrunner, 2017-04-02:
tpacpi.service doesn't run on suspend/resume. tlp-sleep.service does, but doesn't call tpacpi-bat. 

Conclusion: a connection to your problem is highly improbable.

cnzhx, 2017-04-02:
Thank you teleshoes and linrunner for the clarifications. And very sorry for my mistake on jumping to this conclusion.

Maybe I jumped too rush on the conclusion. It's probably the problem just kicked in when I was testing with installing and uninstalling tpacpi-bat. I should've test more times for sure.

But as you explained, it's unlikely the problem is caused by the using of tpacpi-bat. And I ran again into the same problem after I cleaned the install of tpacpi-bat, although only once in several times I tested. It looks like the problem won't happen every time after resuming from suspend.

Thanks again.

linrunner, 2017-04-02:
No problem. 

BTW: you're not alone: https://thinkpad-forum.de/threads/206443-L%C3%BCfter-drehen-nach-Aufwachen-aus-Suspend-auf-Maximum-(Fedora-Core-25)

cnzhx, 2017-04-02:
Hi linrunner, thanks for the new info.