teleshoes/tpacpi-bat #83: Older thinkpad support with coreboot firmware

ArthurHeymans, 2017-08-09:
Hi

Doing some investigation I was able to figure out how to have the EC on my thinkpad X200 set the battery charging thresholds.
Coreboot is a free/open source firmware (BIOS) replacement which means it is possible to modify the ACPI code. The vendor firware has no ACPI method for setting thresholds, but only smapi interface which is not implemented in coreboot.

I was able to implement it in pure ACPI (so no calls to generate SMI to have SMM set the thresholds like most recent thinkpads with vendor firmware) and when doing so I'm able to use tpacpi-bat to set and get thresholds. Inhibit and discharge are not (yet) implemented.
This is the patch: https://review.coreboot.org/#/c/20913/

Do you think its worth adding x200 with coreboot to wiki when this patch is merged as a partially supported device is worthwhile?

Note: it probably also works on thinkpads of the same and older generation.

teleshoes, 2017-08-09:
yea, absolutely! really cool stuff.
(btw, is the X200 one of the ones with the write-protected EC, where you have to use a hardware flasher?)

ArthurHeymans, 2017-08-09:
Ok thanks! We are still a bit debating how to handle the tpacpi-bat compatible interface ACPI (it's not that pretty with bitfields having special meanings instead of multiple arguments), but I think it is possible to create generic tpacpi-bat ACPI code for all coreboot compatible thinkpads (x60/t60, x200/t400 ,x201/t410, x220/t420, x230/t430 and similar).

I think the EC indeed has some sort of password mechanism preventing it's firmware from be written to but that not really something I mess with since I have no clue how to hardware flash the EC.

gch1p, 2018-07-30:
Hi @teleshoes! Did you try to use @ArthurHeymans's x200 patch on your t430s?

ArthurHeymans, 2018-07-30:
https://review.coreboot.org/#/c/coreboot/+/23178/ implements for a handful of thinkpads, but it is still not merged

gch1p, 2018-07-30:
@ArthurHeymans thanks, that's really cool!

teleshoes, 2018-07-30:
(@ArthurHeymans thanks for the update!)

pickfire, 2019-09-07:
https://review.coreboot.org/#/c/coreboot/+/23178/ have been merged

teleshoes, 2020-05-09:
i no longer have a coreboot laptop to test on :(

closed 2020-05-09 by teleshoes