teleshoes/tpacpi-bat #84: Possible to increase minimum 5% before using internal battery?

priccis, 2017-09-04:
First of all, thanks @teleshoes for this script.

So, when using multiple batteries (internal/external) (Power Bridge / Hot Swap), by default the **external** battery will be used till **5%** before the **internal** battery is activated.

Is it possible to implement some way to increase this lower threshold (i.e. use external battery until 10%, and at this point enable the internal)?, or it is controlled at another, lower, level?

teleshoes, 2017-09-04:
:D
yea of course. you just force-discharge on the battery you want to discharge, under whatever circumstances you like. (if you mean, is there a way to tell the BIOS to do this for you, then no. it does whatever ridiculously stupid thing it wants)

my batteries play leapfrog every 10%. to do this, i use tpbattstat (which in turn uses tpacpi-bat). i use it in taffybar, but it also has GTK and dzen2 interfaces. https://github.com/teleshoes/tpbattstat

tpbattstat is more heavyweight than what you need though. here is a script i wrote that may or may not do what you want:
```
#/bin/bash
SWITCH_PCT=10

while [ 1 ]; do
  acOn=`cat /sys/class/power_supply/AC/online`
  intCur=`cat /sys/class/power_supply/BAT0/energy_now`
  intMax=`cat /sys/class/power_supply/BAT0/energy_full`
  extCur=`cat /sys/class/power_supply/BAT1/energy_now`
  extMax=`cat /sys/class/power_supply/BAT1/energy_full`

  intPct=$((100 * intCur / intMax))
  extPct=$((100 * intCur / intMax))

  # reset force-discharge
  tpacpi-bat -s FD 0 1
  tpacpi-bat -s FD 1 1
  tpacpi-bat -s FD 0 0
  tpacpi-bat -s FD 1 0

  if [ $acOn -eq "0" ] && [ $intPct -lt $SWITCH_PCT ] && [ $extPct -gt $SWITCH_PCT ]; then
    echo "$(date) force-discharge external"
    tpacpi-bat -s FD 1 1
  elif [ $acOn -eq "0" ] && [ $intPct -lt $SWITCH_PCT ] && [ $extPct -gt $SWITCH_PCT ]; then
    echo "$(date) force-discharge internal"
    tpacpi-bat -s FD 0 1
  fi

  sleep 60
done
```

closed 2017-09-04 by teleshoes

priccis, 2017-09-06:
Thanks! Will try this

teleshoes, 2017-09-06:
cool. no problem. you can also do the same thing on charge with inhibit-charge

priccis, 2017-10-04:
Didn't have the chance to try it before and it didn't work, but just tried it and realised that force-discharge is not compatible with my T460. :disappointed: