teleshoes/tpacpi-bat #94: T470 + Red Hat Open Client ( RHOC) 7.4 install issue

poutnikl, 2018-02-07:
Hi,

I admit I am rather a novice in any Linux Hw configuration. I usually follow provided step by step cookbooks. If it fails, I am out of luck unless some guide is provided by experienced users.

In recent past I was successful in installing tpacpi-bat management with battery thresholds 40-80 for T430 / RHOC 6.x / Gnome.

When I tried to do the same for my new work laptop T470 / RHOC 7.4 / Gnome  ( As SU cloning the TPACPI-BAT repo,  going to its base folder and running ./install.pl ) , it fails at the make step as below. 

In my undestanding, it should be T470 compatible. But it seems to me the failure is not HW related, but rather OS config related.  Perhaps RHOC 7.3 makes some trouble. Could you advice the proper procedure ?

```
[libor@xxxxxxxxxxx ~]$ su
Password: 
[root@xxxxxxxxxxx libor]# ls
[root@xxxxxxxxxxx libor]# git clone https://github.com/teleshoes/tpacpi-bat
Cloning into 'tpacpi-bat'...
remote: Counting objects: 329, done.
remote: Total 329 (delta 0), reused 0 (delta 0), pack-reused 329
Receiving objects: 100% (329/329), 88.09 KiB | 0 bytes/s, done.
Resolving deltas: 100% (169/169), done.
[root@xxxxxxxxxxx libor]# cd tpacpi-bat
[root@xxxxxxxxxxxx tpacpi-bat]# ls
battery_asl  COPYING  examples  install.pl  README.md  tpacpi-bat
[root@xxxxxxxxxxxx tpacpi-bat]# ./install.pl
sudo cp tpacpi-bat /usr/bin
git clone git://github.com/teleshoes/acpi_call.git /tmp/acpi_call
Cloning into '/tmp/acpi_call'...
remote: Counting objects: 323, done.
remote: Total 323 (delta 0), reused 0 (delta 0), pack-reused 323
Receiving objects: 100% (323/323), 65.84 KiB | 0 bytes/s, done.
Resolving deltas: 100% (150/150), done.
git fetch
git reset HEAD --hard
HEAD is now at 28c3c26 Merge pull request #4 from cefn/master
git clean -fd
git checkout origin/master
Note: checking out 'origin/master'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b new_branch_name

HEAD is now at 28c3c26... Merge pull request #4 from cefn/master
make
make -C /lib/modules/3.10.0-693.11.6.el7.x86_64/build M=/tmp/acpi_call modules
make: *** /lib/modules/3.10.0-693.11.6.el7.x86_64/build: No such file or directory.  Stop.
make: *** [default] Error 2
error running 'make'
[root@xxxxxxxxxxx5 tpacpi-bat]# 
```

poutnikl, 2018-02-07:
I have realized that /lib/modules/3.10.0-693.11.6.el7.x86_64/build  does exist, but its properties says 
link (broken )  (inode/symlink)   with the target  /usr/src/kernels/3.10.0-693.11.6.el7.x86_64.  But there is no item 3.10.0-693.11.6.el7.x86_64 in the folder /usr/src/kernels.

I suppose it may be related with missing kernel sources, but I am not use how to deal with that.

teleshoes, 2018-02-07:
yea, most likely you need to install kernel sources for your (incredibly ancient) kernel. i think you just do `yum install kernel-devel` but redhat is very much not my style.

also, this is a problem with installing one of tpacpi-bat's dependencies, `acpi-call`. its not a problem with installing tpacpi-bat per se.

good luck!

p.s.: latest rhel is based on the latest and greatest of 2013? its a miracle that any modern hardware or software works...

closed 2018-02-07 by teleshoes

poutnikl, 2018-02-08:
Just to provide the feedback - after yum installing of the kernel-devel ( and also missing gcc compiler ) I was successful with the acpi_call installation via install.pl. the set thresholds seem to work now. 

The forced discharging does not, but it seems to be in agreement with the tpacpi-bat legend text.

poutnikl, 2018-03-01:
Correction: Forced discharging FD does work, if just 1 of 2 T470 batteries is chosen. 

 Ignore Charging IC does not work.  tpacpi-bat -g IC 1/2 returns "no" whatever I set by tpacpi-bat -s IC 1/2 0/1