teleshoes/tpacpi-bat #95: Possible to limit discharge on dualbattery setup?

verymilan, 2018-06-03:
Hi there, i am sorry if i misread/missed something, my english experience is making me some trouble in the readme for some reason - however, is it possible to set a discharge limit?
I'd like battery #2 to only discharge to about 20% and then switch over to battery #1 instead of the system default of 5% (as it seems).

Regards

teleshoes, 2018-06-03:
afaik, no, there is no firmware interface for this. i could be wrong of course.

however, it is very easy to implement such a thing in userspace using `tpacpi-bat -s FD` (forceDischarge).
write a simple script that does something like this this:
```
-read B1 and B2 percentages
-if B1 and B2 are both over 20% or both under 20% (or ac is connected), remove FD from both
-if B1 is 20% or less and B2 is over 20%, set FD on B2
```

i do this myself with tpbattstat. (i discharge the batteries 10% at a time, 100 100 => 90 100 => 90 90 => 80 90 => 80 80, etc....)

also, i suggest you check out TLP. i dont use it, so im not sure, but i suspect that this is a feature it has. both TLP and tpbattstat use tpacpi-bat to set the force discharge.

closed 2018-06-03 by teleshoes

verymilan, 2018-06-04:
Alright, thank you very much - it may be not \*that* simple for me as i am not that experienced in scripting though - if you'd share your current script, i \*may* be able to do this - would be very cool.

By the way yesterday when i tried the dynamic service file at bat `#2`at 27%, it did not discharge either really charge - it kept it at 27% - the machine is for a less advanced user who may not be able to force discharge in such a situation, shall i open another ticket for this or may i did something wrong?
I built it from source on Solus on a T450.

teleshoes, 2018-06-04:
what i use is https://github.com/teleshoes/tpbattstat
it does battery balancing during charge+discharge with SMAP, ACPI, or tpacpi-bat, but its principally a battery widget for a system bar.
it has a few different guis (gtk, dzen, json) and can be run in a terminal.

i dont maintain the systemd files, i just host them by request. run tpacpi-bat directly:
```
tpacpi-bat -s FD 1 0 #turn off FD on B1
tpacpi-bat -s FD 2 0 #turn off FD on B2
tpacpi-bat -s FD 2 1 #turn ON FD on B2
```

if FD doesnt work, and it might not, dont bother opening a ticket, theres nothing i can do for you.
if it DOES work, i can probably help you setup tpbattstat to use it.

verymilan, 2018-06-04:
Hmm but as i read tpbattstat depends on tp_smapi which does not support the T450.

Of course i could launch it directly on startup on several ways however i am quite sure it got the flags as supposed - not FD though - more like `-s ST 0 20` and `-s SP 0 90`.
I mean it's late for me already but i did not really understand how you mean your force-discharge examples for a direct run, are you suggesting running that too for ensuring that it doesn't get stuck?
Because as mentioned, a non-console guy is going to use this machine in the future - quite soon actually.

So looks like i gotta start tpacpi-bat again and see if it works fine over a period of time and after he got the machine i gotta experiment with the suggested script (your first answer) - just a bit unfortunate that my T440 does not have dual battery xD

teleshoes, 2018-06-04:
tpbattstat supports both SMAPI and tpacpi-bat. trust me, i wrote it. 

all im saying is that you should just test out whether tpacpi-bat -s FD works.
if it doesnt work manually, then you cant automate it, no matter what other tools you use.

in order to find out if it works, open a terminal. get root (sudo, su, etc).
run this:
```
tpacpi-bat -s FD 1 0
tpacpi-bat -s FD 2 0
tpacpi-bat -s FD 2 1
```
bat#2 should be discharging.

then run this:
```
tpacpi-bat -s FD 1 0
tpacpi-bat -s FD 2 0
tpacpi-bat -s FD 2 1
```
bat#1 should be discharging.

if that works, then you can automate it with tpbattstat, which has a GUI, or write a simple script that runs at startup. if that DOESNT work, youre wasting your time.

teleshoes, 2018-06-04:
p.s.: start and stop charge thresholds are not for balancing between multiple batteries. they are for keeping your battery (or batteries) from charging too much, as leaving batteries fully charge for a long time decreases their lifespan.

verymilan, 2018-06-05:
Thank you very much for the detailed and quick answers! :) 

The good news is, ForceDischarge does work and i was able to run the widget menu and the widget in a window, it does not find a theme engine and seems not to work with Budgie-Desktop tho.
It cannot write to `/proc/acpi/call` - permission denied - i assume i could tweak the sudoers config for that or so? (polkit is installed)
Is there a proper way to launch this in the background whch i didn't find yet?

Regarding your ps, yes i know, making this balancing however would be great.
After a recent attempt it still stuck at >20% at battery2 but later for some reason began to charge before i testwise enforced the discharge - it is confusing, maybe a sort of displaybug/glitch after all?

verymilan, 2018-06-05:
Update: 
On my single-battery T440 it works almost as expected (it charges to the configured max limit but not from the min limit).
The T450 however still freezes the batteries at the current charging state.
This is not about balancing in this case, it's just about using the thresholds.

teleshoes, 2018-06-05:
oh good you use sudo, thats nice. so you can do this:
-add `alias tpacpi-bat="sudo tpacpi-bat"` to your `~/.bashrc`
-add tpacpi-bat to your sudoers with no passwd

alternatively, you can run `sudo tpbattstat -j` in a terminal, and it will balance your batteries once you configure it with `sudo tpbattstat -p`

verymilan, 2018-06-06:
I am slightly confused about why to esp do that to tpacpi-bat yet as i still don't have such a script.^^
However this solution might be useful for tpbattstat.py.

I tested the widget with both balancing interfaces (smapi should be empty though) however the only result yet is "proper" json output but no balancing after all, not while charging, not while discharging. :sob: 
[image]
[image]

teleshoes, 2018-06-06:
hmm, did you run it as root or set up sudoers? its probably just not able to run tpacpi-bat

verymilan, 2018-06-06:
I actually ran it with sudo as advised...
[image]

I only did the sudoers file adjustment for tpacpi-bat yet althrough this may not be needed yet - i mean, it usually seems to freeze the charging status so it's use is somewhat limited. :/
But i gonna try to run it just against one single battery soon, maybe this works.. .

teleshoes, 2018-06-06:
hmm. try doing `sudo bash`, and then checking if `tpacpi-bat` is in your PATH (run `tpacpi-bat` without specifying the full path). then try tpbattstat in that same root terminal

also, it uses inhibit-charge to do balancing while charging (IC, not ST/SP).

verymilan, 2018-06-06:
Yes it is in my PATH, only the applet did not install itself into it as your examples assume, this is why i call the python file directly as in the screenshot above but tpacpi-bat is fine.

teleshoes, 2018-06-06:
is it in your path when you run sudo, is a separate question; there is an option (on by default i think) that overrides your PATH with a limited one

teleshoes, 2018-06-06:
oh i didnt see that edit you wrote. if you have thresholds set, you need to remove those. delete the service file, and manually unset ST and SP.

verymilan, 2018-06-06:
Yes, it is also in the path when i open bash as user from sudo... before i was checking from the bash spawned from sudo su.
This "normal"-user oriented distribution does not come with seperate root password by default afaik.

verymilan, 2018-06-07:
I know i wrote that i may have switched optically but without effect but on recent tries it actually didn't - i even changed the interval to 1 so i can monitor this better - i further modified `battbalance.py` to use the direct path eventho it is definitly in the path fromwhereever i checked - i tried to investigate by understanding the code but i think it's yet too complicated for myself - at least in shortterm (the owner want his laptop this evening) - however it didn't even do the `1 0` -> `0 1` in the output yet.
I even tried using `-w` instead of json output.
[image]
The worst part is that my T440 thresholds with singlebattery appear to be working and i cannot really do further investigating when he got his T450. It's sad.

tgrothe, 2023-12-28:
> what i use is https://github.com/teleshoes/tpbattstat it does battery balancing during charge+discharge with SMAP, ACPI, or tpacpi-bat, but its principally a battery widget for a system bar. it has [...]

Hello @teleshoes !

Is there are possibility to enable/support "force discharge" on this notebook:

```
$ sudo tlp-stat -b
--- TLP 1.6.1 --------------------------------------------

+++ Battery Care
Plugin: thinkpad
Supported features: charge thresholds, recalibration
Driver usage:
* natacpi (thinkpad_acpi) = active (charge thresholds, recalibration)
* tpacpi-bat (acpi_call)  = inactive (superseded by natacpi)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SMP
/sys/class/power_supply/BAT0/model_name                     = 01AV459
/sys/class/power_supply/BAT0/cycle_count                    =     79
/sys/class/power_supply/BAT0/energy_full_design             =  24000 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  21360 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  10730 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Not charging

/sys/class/power_supply/BAT0/charge_control_start_threshold =     40 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =     80 [%]
/sys/class/power_supply/BAT0/charge_behaviour               = [auto] inhibit-charge force-discharge

Charge                                                      =   50.2 [%]
Capacity                                                    =   89.0 [%]

+++ ThinkPad Battery Status: BAT1 (Ultrabay / Slice / Replaceable)
/sys/class/power_supply/BAT1/manufacturer                   = SMP
/sys/class/power_supply/BAT1/model_name                     = 01AV460
/sys/class/power_supply/BAT1/cycle_count                    =    256
/sys/class/power_supply/BAT1/energy_full_design             =  24000 [mWh]
/sys/class/power_supply/BAT1/energy_full                    =  20860 [mWh]
/sys/class/power_supply/BAT1/energy_now                     =  12390 [mWh]
/sys/class/power_supply/BAT1/power_now                      =   3887 [mW]
/sys/class/power_supply/BAT1/status                         = Discharging

/sys/class/power_supply/BAT1/charge_control_start_threshold =     50 [%]
/sys/class/power_supply/BAT1/charge_control_end_threshold   =     60 [%]
/sys/class/power_supply/BAT1/charge_behaviour               = [auto] inhibit-charge force-discharge

Charge                                                      =   59.4 [%]
Capacity                                                    =   86.9 [%]

+++ Charge total                                            =   54.8 [%]
```

I've tried a lot, but `tpacpi-bat -s FD 1 1` does not work, and I cannot switch the battery to be discharged.

I always have to wait until one battery level is at 5% before the other battery is used. Is this something that only depends on the battery controller?

teleshoes, 2023-12-28:
tpacpi-bat is more or less obsolete, except maybe as a debug tool for finding quirky workarounds.
since linux v5.17, thinkpad_acpi kernel module now supports start/stop thresholds and forceDischarge/inihibitCharge, without the need for any userspace software.

instead of: `tpacpi-bat -s FD 1 1`
you can now just do: `echo 'force-discharge' | sudo tee /sys/class/power_supply/BAT0/charge_behaviour`
(and `BAT1` instead of `tpacpi-bat -s FD 2 1`)

if that doesn't work, probably nothing will. good luck!

tgrothe, 2023-12-28:
Thanks for your quick reply, I will try this. :)

tgrothe, 2023-12-29:
Hello, unfortunately this doesn't work with my computer:

```
$ echo 'force-discharge' | sudo tee /sys/class/power_supply/BAT0/charge_behaviour 
force-discharge
tee: /sys/class/power_supply/BAT0/charge_behaviour: Eingabe-/Ausgabefehler
```

There is an input/output error each time. But thanks for your effort. :)