teleshoes/tpacpi-bat #96: Battery Capacity Is Wrongly Reported After Running tpacpi-bat

emariz, 2018-08-17:
I recently purchased a Thinkpad E470 and this week I installed and configured tpacpi-bat.
The system is running Debian Testing.
`4.17.0-1-amd64 #1 SMP Debian 4.17.8-1 (2018-07-20) x86_64 GNU/Linux`

On the one had, tpacpi-bat works, for it can correctly set charging thresholds on this battery. On the other, the battery design capacity is wrongly reported (by acpi) after running it.

The design capacity of the battery on this system is 45280 mWh, and until this week acpi had been reporting a close capacity (around 4200 mAh.) After installing tpacpi-bat and setting the thresholds (40 and 80 %), now acpi only reports a capacity 'capped' by them, even after resetting them to zero.
```
$ acpi --everything
Battery 0: Discharging, 100%, discharging at zero rate - will never fully discharge.
Battery 0: design capacity 3621 mAh, last full capacity 3761 mAh = 100%
Adapter 0: on-line
```
That is the output after resetting the thresholds and restarting the system.
The now-displayed design capacity (3621 mAh) is lower than what it was displayed earlier this week (around 4200 mAh.) Moreover, earlier this morning (when I noticed the issue), acpi was reporting a capacity of around 3900 mAh.

I first thought that this was the expected behaviour, that to inhibit charging one had to 'fake' the capacity. But the sudden decrease in the capacity reports without altering the thresholds lead me to think otherwise.

Granted, acpi is a different tool and it may be misreading the values, but these values only changed after using tpacpi-bat.

Nothing extraneous happened to the battery (or the laptop) within this week that could have suddenly affected the battery life. The laptop itself has barely been used since it was purchased, and I bet that the amount of time that the battery has been in use does not even account for two full discharges. It is that new.

I already tried removing the AC cable and the battery, then holding the Power button for 30 seconds and the pushing it thirty times consecutively, but the design capacity is still reported as around 3600 mAh.

Any leads on what might have happened?

Some information from dmidecode.
```
System Information
	Manufacturer: LENOVO
	Product Name: 20H1A01XAC
	Version: ThinkPad E470

BIOS Information
	Vendor: LENOVO
	Version: R0DET95W (1.95 )
	Release Date: 06/16/2018

Portable Battery
	Location: Front
	Manufacturer: SMP
	Name: 01AV413
	Design Capacity: 45280 mWh
	Design Voltage: 11100 mV
	SBDS Version: 00.00
	Maximum Error: Unknown
	SBDS Serial Number: 1C84
	SBDS Manufacture Date: 2017-09-30
	SBDS Chemistry: LiP
	OEM-specific Information: 0x00000000
```

teleshoes, 2018-08-17:
.nope! if it was only the capacity being reported as 3600, i would suggest that the battery did in fact suffer trauma unknown to you, or just coincidentally took a dive. the fact that its ALSO misreading the design capacity suggests a bug in the driver.

i suggest you keep trying to set all the values to the defaults a few times, in different ways, and also see if you can make the problem worse (slightly lower thresholds, perhaps) and then make it better again, just to verify that tpacpi-bat is in fact affecting the way acpi is reading the values.

also, on a side note, the kernel module thinkpad_acpi changed in 4.17, adding kernel modules for start/stop thresholds. you might wanna try those, though they should behave the same as tpacpi-bat.

good luck!

closed 2018-08-17 by teleshoes

emariz, 2018-08-21:
Some data after a couple of days of testing the battery after resetting the values with tpacpi-bat.

- Somehow, the values of battery's design capacity and full capacity increase as the battery discharges (when the batty is almost depleted, those values are as high as they could be.)
```
$ cat /sys/class/power_supply/BAT0/uevent 

POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_STATUS=Discharging
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_TECHNOLOGY=Li-poly
POWER_SUPPLY_CYCLE_COUNT=0
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=11100000
POWER_SUPPLY_VOLTAGE_NOW=11184000
POWER_SUPPLY_POWER_NOW=4618000
POWER_SUPPLY_ENERGY_FULL_DESIGN=45280000
POWER_SUPPLY_ENERGY_FULL=47390000
POWER_SUPPLY_ENERGY_NOW=19540000
POWER_SUPPLY_CAPACITY=41
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_MODEL_NAME=01AV413
POWER_SUPPLY_MANUFACTURER=SMP
POWER_SUPPLY_SERIAL_NUMBER= 7300
```
```
$ acpi --everything
Battery 0: Discharging, 63%, 05:28:22 remaining
Battery 0: design capacity 3961 mAh, last full capacity 4145 mAh = 100%
Adapter 0: off-line

$ acpi --everything
Battery 0: Discharging, 41%, 04:16:19 remaining
Battery 0: design capacity 4048 mAh, last full capacity 4236 mAh = 100%
Adapter 0: off-line

$ acpi --everything
Battery 0: Discharging, 27%, 03:14:57 remaining
Battery 0: design capacity 4045 mAh, last full capacity 4234 mAh = 100%
Adapter 0: off-line

$ acpi --everything
Battery 0: Discharging, 10%, 01:15:06 remaining
Battery 0: design capacity 4307 mAh, last full capacity 4507 mAh = 100%
Adapter 0: off-line

$ acpi --everything
Battery 0: Discharging, 2%, 00:19:24 remaining
Battery 0: design capacity 4507 mAh, last full capacity 4638 mAh = 100%
Adapter 0: off-line
```
- On the other hand, the values of design capacity and full capacity decrease as the battery charges.
```
$ acpi --everything
Battery 0: Charging, 27%, 01:04:46 until charged
Battery 0: design capacity 3821 mAh, last full capacity 3859 mAh = 100%
Adapter 0: on-line

$ acpi --everything
Battery 0: Charging, 54%, 00:39:14 until charged
Battery 0: design capacity 3736 mAh, last full capacity 3774 mAh = 100%
Adapter 0: on-line

$ acpi --everything
Battery 0: Charging, 86%, 00:16:06 until charged
Battery 0: design capacity 3631 mAh, last full capacity 3668 mAh = 100%
Adapter 0: on-line

$ acpi --everything
Battery 0: Charging, 90%, 00:14:25 until charged
Battery 0: design capacity 3624 mAh, last full capacity 3660 mAh = 100%
Adapter 0: on-line

$ acpi --everything
Battery 0: Discharging, 100%, discharging at zero rate - will never fully discharge.
Battery 0: design capacity 3624 mAh, last full capacity 3781 mAh = 100%
Adapter 0: on-line
```
- And discharging again...
```
$ acpi --everything
Battery 0: Discharging, 94%, 06:46:54 remaining
Battery 0: design capacity 3723 mAh, last full capacity 3881 mAh = 100%
Adapter 0: off-line

$ acpi --everything
Battery 0: Discharging, 67%, 05:09:41 remaining
Battery 0: design capacity 3924 mAh, last full capacity 4090 mAh = 100%
Adapter 0: off-line
```

So far, the battery seems to last the expected time, but capacity values change as it charges or discharges. This did not happened before tpacpi-bat was installed.

Note that the highest values for design capacity and full capacity (4507 mAh and 4638 mAh, respectively) are about 20 % higher than their minimums (3624 mAh and 3781 mAh, respectively.) And this coincides with the 80 % threshold cap originally set with tpacpi-bat.

teleshoes, 2018-08-22:
interesting. that MIGHT simply mean that your battery's voltage is unpredictable, presuming that amps are calculated. anyway, it sounds like its a bug in acpi. you should check the kernel file devices as well

emariz, 2018-08-27:
Some more reads.

```
$ ls -1 /sys/class/power_supply/BAT0/
alarm
capacity
capacity_level
charge_start_threshold
charge_stop_threshold
cycle_count
device
energy_full
energy_full_design
energy_now
manufacturer
model_name
power
power_now
present
serial_number
status
subsystem
technology
type
uevent
voltage_min_design
voltage_now
```
- Discharging, at 61 %.
```
$ acpi --everything
Battery 0: Discharging, 61%, 05:43:30 remaining
Battery 0: design capacity 3972 mAh, last full capacity 4087 mAh = 100%
Adapter 0: off-line

$ cat /sys/class/power_supply/BAT0/capacity
61

$ cat /sys/class/power_supply/BAT0/energy_full
46590000

$ cat /sys/class/power_supply/BAT0/energy_full_design 
45280000

$ cat /sys/class/power_supply/BAT0/energy_now 
23600000

$ cat /sys/class/power_supply/BAT0/power_now
5156000
```
- Discharging, at 15 %.
```
$ acpi --everything
Battery 0: Discharging, 15%, 00:58:22 remaining
Battery 0: design capacity 4229 mAh, last full capacity 4351 mAh = 100%
Adapter 0: off-line

$ cat /sys/class/power_supply/BAT0/capacity
15

$ cat /sys/class/power_supply/BAT0/energy_full
46590000

$ cat /sys/class/power_supply/BAT0/energy_full_design
45280000

$ cat /sys/class/power_supply/BAT0/energy_now 
6930000

$ cat /sys/class/power_supply/BAT0/power_now 
5022000
```
- Discharging, at 10 %.
I removed 'energy_full' and 'energy_full_design' since they are fixed values.
I also noticed that 'power_now' varied from one read to another. I posted the maximum and minimum out of five reads.
```
$ acpi --everything
Battery 0: Discharging, 10%, 01:04:34 remaining
Battery 0: design capacity 4303 mAh, last full capacity 4427 mAh = 100%
Adapter 0: off-line

$ cat /sys/class/power_supply/BAT0/capacity
10

$ cat /sys/class/power_supply/BAT0/energy_now 
4660000

$ cat /sys/class/power_supply/BAT0/power_now 
4346000

$ cat /sys/class/power_supply/BAT0/power_now 
4817000
```
- Discharging, at 4 %.
I also posted a couple of consecutive 'power_now' reads.
```
$ acpi --everything
Battery 0: Discharging, 4%, 00:26:35 remaining
Battery 0: design capacity 4368 mAh, last full capacity 4495 mAh = 100%
Adapter 0: off-line

$ cat /sys/class/power_supply/BAT0/capacity
4

$ cat /sys/class/power_supply/BAT0/energy_now 
1930000

$ cat /sys/class/power_supply/BAT0/power_now 
4357000

$ cat /sys/class/power_supply/BAT0/power_now 
4402000

$ cat /sys/class/power_supply/BAT0/power_now 
4196000
```
- Discharging, at 3 %.
This read was done just before the system entered hibernation automatically.
As on my previous posts, acpi shows the greatest design- and full- capacities when the battery is almost empty.
```
$ acpi --everything
Battery 0: Discharging, 3%, 00:24:32 remaining
Battery 0: design capacity 4377 mAh, last full capacity 4504 mAh = 100%
Adapter 0: off-line

$ cat /sys/class/power_supply/BAT0/energy_now 
1620000

$ cat /sys/class/power_supply/BAT0/power_now 
4407000

$ cat /sys/class/power_supply/BAT0/power_now 
4190000
```
- After a full charge.
'enery_now' seems to show the "correct" full capacity value, while the value of 'power_now' is now zero.
```
$ acpi --everything
Battery 0: Discharging, 100%, discharging at zero rate - will never fully discharge.
Battery 0: design capacity 3623 mAh, last full capacity 3772 mAh = 100%
Adapter 0: on-line

$ cat /sys/class/power_supply/BAT0/energy_now 
47140000

$ cat /sys/class/power_supply/BAT0/power_now 
0
```