teleshoes/tpacpi-bat #97: <bat> missing or incorrect

kinkod, 2019-05-05:
when i run **sudo service tpacpi start**, it fails. The error message is "<bat> missing or incorrect". I have tried setting battery to 0 1 and 2 in config file. Same error with all numbers. Also removed doublequotes so the numbers are integers, same problem.

System is Lenovo ThinkPad Edge E530 which is listed as a supported system.
Running Linux Mint 19.1

**/etc/conf.d/tpacpi/tpacpi.conf.d**
`# battery is 1 for main, 2 for secondary, or 0 for either/both
BATTERY="0"

# Start charging threshold (0 for default, 1-99 for percent)
START_THRESHOLD="40"

# Stop charging threshold (0 for default, 1-99 for percent)
STOP_THRESHOLD="80"
`
**journalctl -xe**
`-- Unit tpacpi.service has begun starting up.
May 05 07:20:15 Carbon tpacpi-bat[18785]: <bat> missing or incorrect
May 05 07:20:15 Carbon systemd[1]: tpacpi.service: Main process exited, code=exited, status=255/n/a
May 05 07:20:15 Carbon systemd[1]: tpacpi.service: Failed with result 'exit-code'.
May 05 07:20:15 Carbon systemd[1]: Failed to start sets battery thresholds.
-- Subject: Unit tpacpi.service has failed
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
`

teleshoes, 2019-05-05:
sorry, the service files are provided only for convenience and is not supported (by me) at all.
if you run `tpacpi-bat` directly, does it work?

e.g.:
```
wolke:~$ sudo tpacpi-bat -s ST 1 0
wolke:~$ sudo tpacpi-bat -g ST 1
0 (default)
wolke:~$ sudo tpacpi-bat -s ST 1 50
wolke:~$ sudo tpacpi-bat -g ST 1
50 (relative percent)
wolke:~$ sudo tpacpi-bat -s ST 1 0
wolke:~$ sudo tpacpi-bat -g ST 1
0 (default)
```

kinkod, 2019-05-07:
Seems to work. Thank you.

sudo tpacpi-bat -s ST 1 0

sudo tpacpi-bat -s SP 1 0

These reset to default?

kinkod, 2019-05-07:
sudo tpacpi-bat -g ST 1
128 (unknown)

Why does it show 128 (unknown) instead of 0 (default)? Any idea?

teleshoes, 2020-05-09:
(sorry! very likely #37. which in turn is likely a firmware bug)

closed 2020-05-09 by teleshoes