teleshoes/tpacpi-bat #98: Start charge threshold does not work

372046933, 2019-05-26:
Below is my configuration.
```
# battery is 1 for main, 2 for secondary, or 0 for either/both
BATTERY="0"

# Start charging threshold (0 for default, 1-99 for percent)
START_THRESHOLD="40"

# Stop charging threshold (0 for default, 1-99 for percent)
STOP_THRESHOLD="80"
```
And systemd as follows:
```
ExecStart=/usr/bin/tpacpi-bat -s ST $BATTERY $START_THRESHOLD
ExecStart=/usr/bin/tpacpi-bat -s SP $BATTERY $STOP_THRESHOLD
ExecStop=/usr/bin/tpacpi-bat -s ST $BATTERY 0
ExecStop=/usr/bin/tpacpi-bat -s SP $BATTERY 0
```
In my situation, only ST threshold is effective. Each time I plug the power, the battery get charged to 80%. But the configuration says it should charge when below 40%. The laptop is Thinkpad X1 6th Gen.

teleshoes, 2019-05-26:
try this:
`tpacpi-bat -g ST 0` and `tpacpi-bat -g SP 0`

if those return 40 and 80, then `tpacpi-bat` is doing everything it should do, and whatever behavior your laptop does is all you can expect. sorry, but the behavior is inside the firmware.

however, if the getters DONT work, its possible something is wrong with your setup. try setting battery = 1, for example. some firmwares do annoyingly different things (also, of course, while debugging, stop the systemd service and run the `-s` and `-g` commands yourself)

also test FD and IC. if those work, then you can technically implement the behavior you want in userspace with a script that does the following:
```
-monitor AC attached and battery % every 10s (or 1s or 60s etc)
-if AC status is still the same as last check (still plugged or still unplugged), do nothing
-if AC is newly attached this check and battery < 40, remove force discharge
-if AC is newly attached this check  and battery is between 40 and 80, set force discharge
-if AC is newly attached this check  and battery > 80, set inhibit charge
-if AC is newly UNattached this check , remove force discharge and inhibit charge
```

teleshoes, 2019-05-26:
also, i had no idea tpacpi-bat worked on last years X1c, at all. good to know, thanks!

372046933, 2019-05-26:
Thanks a million. I guess the problem is about the battery index. When I run `sudo tpacpi-bat -v -g ST 0` or `sudo tpacpi-bat -v -g ST 0`. Both commands returns error like `Cannot specify 'either/both' for reading ST
`

372046933, 2019-05-26:
I used `BATTERY="1"` in systemd, but it is still charging when battery level is above 40%

linrunner, 2019-05-26:
@372046933 : the service is unnecessary. Just set the required thresholds once via command line and they will persist (stored in the EC), unless you remove the battery, which isn't that easy with the X1C6. You'll also want to update your BIOS to the newest version with EC firmware 1.13 for the upper threshold to work properly ([link](https://linrunner.de/en/tlp/docs/tlp-faq.html#stop-threshold-sticks)). Btw: why don't you just use TLP?

@teleshoes : yes, all newer ThinkPads seem to work well with tpacpi-bat – and the same ACPI calls in the kernel (natacpi) too. With the usual peculiarities for E and L series of course ...

372046933, 2019-05-27:
@linrunner : I is using 1.13 firmware now. It seems that power adapters have something to do with battery level. When I use adapter of Pixel 2 XL, X1C6 just ignored the start charge threshold. But battery now stays at 72% when using X1's original power adapter.

372046933, 2019-05-27:
@linrunner  Do you mean that the configuration stays in EC and works in Windows too? I have Windows 10 dual boot.

372046933, 2019-05-28:
Now I can confirm that `tpacpi-bat` works only when the adapter is plugged **after** Linux is booted. If the power is plugged before booting, it will continue to charge the battery until 80%. IMHO, EC does not depend on OS. So maybe the configuration is stored on OS.

linrunner, 2019-05-28:
> When I use adapter of Pixel 2 XL, X1C6 just ignored the start charge threshold.
A smartphone charger. Are you kidding?

The charge thresholds work even when the machine is off. No OS involved.

372046933, 2019-06-02:
@linrunner  The power of Pixel 2 XL adapter is 18W(9v * 2A). It can be utilized to supply the X6C under light workload. By the way, the adapter of X16C can charge Pixel 2 XL.
On heavy workload, smart phone charger is not enough because the battery level continues to drop. In my situation, the configure of threshold does not work in Windows. It is only effective when Linux is booted and the adapter is plugged in after that.

linrunner, 2019-06-02:
A discussion about funny symptoms with an underspecced charger doesn't make sense to me. Good luck.

372046933, 2019-06-02:
There is a misunderstanding here. It doesn't have to do with adapter, but how the tpacpi works. I can confirm that the threshold does not work in Windows no matter which adapter which is used.

closed 2020-05-09 by teleshoes