teleshoes/tpacpi-bat #104: raising ST and SP after another results in ST not changing as expected

henning-schild, 2021-03-25:
This might be releated to #37 but i am not sure. Seen on an X280 with a 5.4.80 kernel and acpi_call 1.2.1.

Steps to reproduce:

lowering both values works
```
 /tmp # tpacpi-bat -g ST 1; tpacpi-bat -g SP 1
62 (relative percent)
82 (relative percent)
 /tmp # tpacpi-bat -s ST 1 32; tpacpi-bat -s SP 1 42
 /tmp # tpacpi-bat -g ST 1; tpacpi-bat -g SP 1
32 (relative percent)
42 (relative percent)
```

raising needs to be done twice
```
 /tmp # tpacpi-bat -s ST 1 62; tpacpi-bat -s SP 1 82
 /tmp # tpacpi-bat -g ST 1; tpacpi-bat -g SP 1
32 (relative percent)
82 (relative percent)
 /tmp # tpacpi-bat -s ST 1 62; tpacpi-bat -s SP 1 82
 /tmp # tpacpi-bat -g ST 1; tpacpi-bat -g SP 1
62 (relative percent)
82 (relative percent)
```

going back down works as expected

Turning the raise around works as expected
```
tpacpi-bat -s SP 1 82; tpacpi-bat -s ST 1 62
```
But in that order the lowering is affected

```
 /tmp # tpacpi-bat -g ST 1; tpacpi-bat -g SP 1
62 (relative percent)
82 (relative percent)
 /tmp # tpacpi-bat -s SP 1 42; tpacpi-bat -s ST 1 32
 /tmp # tpacpi-bat -g ST 1; tpacpi-bat -g SP 1
32 (relative percent)
82 (relative percent)
```

https://github.com/gentoo/gentoo/pull/12121

henning-schild, 2021-03-25:
tried on an X230, that one is not affected

henning-schild, 2021-03-25:
Setting "start, stop, start" or "stop, start, stop" as opposed to "star, stop" or "stop, start" allows a stable raise and lowering

henning-schild, 2021-03-25:
> Setting "start, stop, start" or "stop, start, stop" as opposed to "star, stop" or "stop, start" allows a stable raise and lowering

https://github.com/gentoo/gentoo/pull/12121#issuecomment-806511857

linrunner, 2021-03-25:
Imho this is not a tpacpi-bat issue but likely a EC firmware bug on the X280. Is your BIOS / EC firmware up to date?

TLP 1.4 will contain a change for reading a threshold twice:

- https://github.com/linrunner/TLP/issues/369
- https://github.com/linrunner/TLP/commit/98e34f82daae9d4b754d46f44399714b7386fa84

And its not the only issue with Lenovo's firmwares of the *80 generation: https://linrunner.de/tlp/faq/battery.html#disabling-the-charge-thresholds-does-not-work

henning-schild, 2021-03-26:
> Imho this is not a tpacpi-bat issue but likely a EC firmware bug on the X280. Is your BIOS / EC firmware up to date?

installed latest x280 BIOS "1.41" and the problem persists

teleshoes, 2021-03-26:
drat. anyway, i cant imagine any way this could possibly be a tpacpi-bat bug. i also dont think the double-run workaround for the EC bug belongs in tpacpi-bat, although a case could be made for it. closing this, feel free to re-open if you feel strongly that the workaround belongs in tpacpi-bat and not in the calling application (e.g. tlp)

closed 2021-03-26 by teleshoes

henning-schild, 2021-03-27:
I feel tpacpi-bat would be the better place, since it is more central. Not sure how many people know or use tlp, i see no need for it with what gentoo offers in simple scripts shipped with tpacpi-bat ... the place i discovered the problem and did not want to just fix it there.
https://github.com/gentoo/gentoo/pull/12121

linrunner, 2021-03-27:
Best thing would be if Lenovo would fix its firmware. Try to rise the issue at Lenovo forums, maybe @MarkRHPearson takes care of it. Good luck.

Btw: the native kernel interface, i.e. 

    /sys/class/power_supply/BAT0/charge_control_start_threshold
    /sys/class/power_supply/BAT0/charge_control_end_threshold

will also suffer from this bug because it employs exactly the same ACPI calls as tpacpi-bat does.

teleshoes, 2021-03-27:
i see what youre saying, that it would be convenient to have it in one central place, but centrality isnt the issue. the issue is whether the low-level tool should add quirky workarounds for faulty firmware, or if that should be left to the caller. since tpacpi-bat strives to be as thin a wrapper around the ACPI interface as possible, i think it would be odd to make 2 acpi calls transparently.

henning-schild, 2021-03-27:
@linrunner how about i will repair the gentoo side of tlp and you see how to get the attention of  @MarkRHPearson, not sure i want to sign up to yet another forum ...

Ah well, i guess i will look into tlp and its gentoo bits anyways, no strings attached