teleshoes/tpacpi-bat #105: kernel NULL pointer dereference on Linux kernel 5.13

ndoo, 2021-07-28:
`uname -a` output:
`Linux doopad-x395 5.13.4-200.fc34.x86_64 #1 SMP Tue Jul 20 20:27:29 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`

`dmesg` output:

```
[45420.141212] BUG: kernel NULL pointer dereference, address: 0000000000000000
[45420.141217] #PF: supervisor instruction fetch in kernel mode
[45420.141220] #PF: error_code(0x0010) - not-present page
[45420.141221] PGD 0 P4D 0 
[45420.141224] Oops: 0010 [#4] SMP NOPTI
[45420.141226] CPU: 3 PID: 85578 Comm: tpacpi-bat Tainted: G      D    O      5.13.4-200.fc34.x86_64 #1
[45420.141229] Hardware name: LENOVO 20NLCTO1WW/20NLCTO1WW, BIOS R13ET49P(1.23 ) 11/24/2020
[45420.141231] RIP: 0010:0x0
[45420.141236] Code: Unable to access opcode bytes at RIP 0xffffffffffffffd6.
[45420.141238] RSP: 0018:ffffa60ec2fc3ef0 EFLAGS: 00010246
[45420.141240] RAX: 0000000000000000 RBX: ffff93c9e1c8af00 RCX: 0000000000000004
[45420.141241] RDX: 0000000000000001 RSI: 0000000000000000 RDI: ffff93ca260cbc00
[45420.141243] RBP: 0000000000000000 R08: 0000000000000003 R09: 0000000000000000
[45420.141244] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000000001
[45420.141245] R13: ffffffffffffffea R14: ffff93ca260cbc00 R15: 0000000000000000
[45420.141246] FS:  00007f1ace0b7740(0000) GS:ffff93cc70ac0000(0000) knlGS:0000000000000000
[45420.141248] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[45420.141250] CR2: ffffffffffffffd6 CR3: 000000013740e000 CR4: 00000000003506e0
[45420.141251] Call Trace:
[45420.141253]  proc_reg_llseek+0x49/0x80
[45420.141259]  ? __fdget_pos+0x13/0x50
[45420.141263]  ksys_lseek+0x7d/0xb0
[45420.141267]  do_syscall_64+0x40/0x80
[45420.141271]  entry_SYSCALL_64_after_hwframe+0x44/0xae
[45420.141275] RIP: 0033:0x7f1ace35341b
[45420.141276] Code: ff ff c3 0f 1f 40 00 48 8b 15 59 1a 0d 00 f7 d8 64 89 02 48 c7 c0 ff ff ff ff eb ba 0f 1f 00 f3 0f 1e fa b8 08 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 05 c3 0f 1f 40 00 48 8b 15 29 1a 0d 00 f7 d8
[45420.141278] RSP: 002b:00007fff7b53a158 EFLAGS: 00000246 ORIG_RAX: 0000000000000008
[45420.141281] RAX: ffffffffffffffda RBX: 00007f1ace7b4b20 RCX: 00007f1ace35341b
[45420.141282] RDX: 0000000000000001 RSI: 0000000000000000 RDI: 0000000000000003
[45420.141283] RBP: 000055d4dec0c560 R08: 00007fff7b53a100 R09: 00000000ffffffff
[45420.141284] R10: 00007f1ace425a00 R11: 0000000000000246 R12: 000055d4debeb2a0
[45420.141285] R13: 00007fff7b53a360 R14: 000055d4debeb3f0 R15: 000055d4decbf590
[45420.141288] Modules linked in: ipheth apple_mfi_fastcharge hid_sony ff_memless acpi_call(O) snd_usb_audio snd_usbmidi_lib snd_rawmidi uinput rfcomm snd_seq_dummy snd_hrtimer xt_CHECKSUM xt_MASQUERADE xt_conntrack ipt_REJECT nf_nat_tftp nf_conntrack_tftp bridge stp llc nft_objref nf_conntrack_netbios_ns nf_conntrack_broadcast nft_fib_inet nft_fib_ipv4 nft_fib_ipv6 nft_fib nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject nft_ct nft_chain_nat ip6table_nat ip6table_mangle ip6table_raw ip6table_security iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev mc iptable_mangle iptable_raw iptable_security ip_set bnep nf_tables nfnetlink ip6table_filter ip6_tables iptable_filter btusb btrtl btbcm btintel bluetooth iwlmvm sunrpc mac80211 snd_ctl_led intel_rapl_msr intel_rapl_common snd_hda_codec_realtek libarc4 snd_hda_codec_generic snd_hda_codec_hdmi edac_mce_amd snd_hda_intel kvm_amd
[45420.141333]  snd_intel_dspcfg vfat snd_intel_sdw_acpi kvm fat snd_hda_codec iwlwifi snd_hda_core snd_hwdep irqbypass snd_seq rapl pcspkr snd_seq_device cfg80211 wmi_bmof joydev snd_rn_pci_acp3x snd_pcm thinkpad_acpi k10temp i2c_piix4 snd_pci_acp3x platform_profile ledtrig_audio snd_timer snd ipmi_devintf ucsi_acpi soundcore ipmi_msghandler typec_ucsi typec rfkill i2c_scmi acpi_cpufreq zram ip_tables dm_crypt trusted asn1_encoder amdgpu rtsx_pci_sdmmc mmc_core drm_ttm_helper ttm iommu_v2 gpu_sched i2c_algo_bit drm_kms_helper crct10dif_pclmul crc32_pclmul crc32c_intel cec ghash_clmulni_intel drm serio_raw sp5100_tco ccp nvme rtsx_pci nvme_core r8169 wmi video fuse
[45420.141372] CR2: 0000000000000000
[45420.141373] ---[ end trace d2070d80e91f6c32 ]---
[45420.141375] RIP: 0010:0x0
[45420.141377] Code: Unable to access opcode bytes at RIP 0xffffffffffffffd6.
[45420.141378] RSP: 0018:ffffa60ec251bef0 EFLAGS: 00010246
[45420.141379] RAX: 0000000000000000 RBX: ffff93c9e1c8af00 RCX: 0000000000000001
[45420.141381] RDX: 0000000000000001 RSI: 0000000000000000 RDI: ffff93ca844d7b00
[45420.141382] RBP: 0000000000000000 R08: 0000000000000003 R09: 0000000000000000
[45420.141383] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000000001
[45420.141384] R13: ffffffffffffffea R14: ffff93ca844d7b00 R15: 0000000000000000
[45420.141385] FS:  00007f1ace0b7740(0000) GS:ffff93cc70ac0000(0000) knlGS:0000000000000000
[45420.141387] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[45420.141388] CR2: ffffffffffffffd6 CR3: 000000013740e000 CR4: 00000000003506e0
```

Hardware:
Lenovo ThinkPad X395 (20NL/20NLCTO1WW)

`tlp-stat -b` output:
```
/usr/share/tlp/func.d/35-tlp-func-batt: line 25: 86685 Killed                  $TPACPIBAT -g FD 1 > /dev/null 2>&1
--- TLP 1.3.1 --------------------------------------------

+++ Battery Features: Charge Thresholds and Recalibrate
natacpi    = active (data, thresholds)
tpacpi-bat = unknown status
tp-smapi   = inactive (ThinkPad not supported)

/usr/share/tlp/func.d/35-tlp-func-batt: line 25: 86731 Killed                  $TPACPIBAT -g FD 1 > /dev/null 2>&1
/usr/share/tlp/func.d/35-tlp-func-batt: line 25: 86777 Killed                  $TPACPIBAT -g FD 1 > /dev/null 2>&1
+++ ThinkPad Battery Status: BAT0
/sys/class/power_supply/BAT0/manufacturer                   = SMP
/sys/class/power_supply/BAT0/model_name                     = 02DL018
/sys/class/power_supply/BAT0/cycle_count                    =     95
/sys/class/power_supply/BAT0/energy_full_design             =  48100 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  44570 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  22080 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Unknown

/sys/class/power_supply/BAT0/charge_start_threshold         =     30 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold          =     50 [%]

Charge                                                      =   49.5 [%]
Capacity                                                    =   92.7 [%]
```

I cross-posted this issue at [acpi_call](https://github.com/mkottman/acpi_call/issues/96) as well.

andrew-d, 2021-08-19:
I just ran into this as well on [NixOS](https://nixos.org/); if anyone else does, you can use the following commit (or as an overlay) to fix it:
https://github.com/andrew-d/nixpkgs/commit/83d3fa0aa1c363c163efd64cefc13f27d4bb135b

roadkell, 2021-10-21:
I have the same bug on Thinkpad T430 with Pop!_OS. After some research, I've written a gist on fixing it manually on your local machine (in case your distro has an outdated version of `acpi_call`): https://gist.github.com/roadkell/9e98db6656e28fbbf1bf51082040f67f

teleshoes, 2023-11-15:
(closing as acpi_call dependency/package-management issues are outside of scope of this repo)

closed 2023-11-15 by teleshoes