teleshoes/tpacpi-bat #110: nixos pkgs.tpacpi-bat doesn't install acpi_call

markifi, 2023-08-29:
`[user@nixos:~]$ tpacpi-bat -s ST 1 86modprobe: FATAL: Module acpi_call not found in directory /run/booted-system/kernel-modules/lib/modules/6.1.45
Could not find /proc/acpi/call. Is module acpi_call loaded?
`

shouldn't it install that too? especially as acpi_call by itself is not in nixpkgs

HoseanRC, 2023-11-15:
you have to install acpi_call (or acpi_call-dkms if you use a kernel other than linux) through your package manager
if you have debian based distro, do `sudo apt install acpi_call`
if you have arch based distro, do `sudo pacman -S acpi_call` or `yay -S acpi_call` if you use yet another yogurt and so on...

teleshoes, 2023-11-15:
thanks. yea, dependency/package-management issues are a little out of scope for this repo (which is really just a perl script). good luck!

closed 2023-11-15 by teleshoes